gazebosim/gazebo_test_cases #1679: gz-sim: spacecraft.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/spacecraft.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
interested

bperseghetti, 2024-08-29:
I can test as well.

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 14-31-33.webm](https://github.com/user-attachments/assets/beb67ba5-dc1e-460c-92e5-a80057ffa780)
LGTM, odom and thrusters worked as expected

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * ```spacecraft.sdf``` world opens successfully with no issue. Space craft changes attitude when ```gz topic -p 'normalized:[1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]' -t /dart/command/duty_cycle --msgtype gz.msgs.Actuators``` is passed and odometry is updated accordingly.

[image]

* **Issues**
    * N/A
    
* **Fixes**
    * N/A

* **Comments/Suggestions**
    * N/A

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde