gazebosim/gazebo_test_cases #1683: gz-sim: touch_plugin.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/touch_plugin.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-30:
Interested

Combuster54, 2024-08-31:
## Ubuntu 24.04 Noble

Running touch_plugin.sdf
```bash
gz sim touch_plugin.sdf
```
[image]

touch plugin is able by default 

[image]

after certain time in contact, plugin sends true

## Conclusion
touch plugin works as expected

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde