gazebosim/gazebo_test_cases #1684: gz-sim: track_drive.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/track_drive.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-30:
Interested

khshmt, 2024-08-30:
## Ubuntu 22.04 (jammy)
[Screencast from 08-31-2024 12:32:37 AM.webm](https://github.com/user-attachments/assets/65ea2661-ce71-40ed-b289-a6f8d2a22068)

azeey, 2024-08-30:
I noticed you used Jammy. Can you confirm you used Gazebo Ionic? You can find from the about dialog in the [hamberger menu](https://en.wikipedia.org/wiki/Hamburger_button)?

khshmt, 2024-08-30:
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde