gazebosim/gazebo_test_cases #1690: gz-sim: tunnel.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/tunnel.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-30:
Interested

Combuster54, 2024-08-30:
### Ubuntu 20.04 Noble

Ok. I was trying to run this world and I find some issues.

```bash
combuster@combuster54:~$ gz sim tunnel.sdf
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_front_wheel_link"]/collision[@name="left_front_wheel_link_collision"]/surface/friction/ode/max_vel:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1300]: XML Element[max_vel], child of element[ode], not defined in SDF. Copying[max_vel] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_front_wheel_link"]/collision[@name="left_front_wheel_link_collision"]/surface/friction/ode/min_depth:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1301]: XML Element[min_depth], child of element[ode], not defined in SDF. Copying[min_depth] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_rear_wheel_link"]/collision[@name="left_rear_wheel_link_collision"]/surface/friction/ode/max_vel:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1383]: XML Element[max_vel], child of element[ode], not defined in SDF. Copying[max_vel] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_rear_wheel_link"]/collision[@name="left_rear_wheel_link_collision"]/surface/friction/ode/min_depth:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1384]: XML Element[min_depth], child of element[ode], not defined in SDF. Copying[min_depth] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="right_front_wheel_link"]/collision[@name="right_front_wheel_link_collision"]/surface/friction/ode/max_vel:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1466]: XML Element[max_vel], child of element[ode], not defined in SDF. Copying[max_vel] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="right_front_wheel_link"]/collision[@name="right_front_wheel_link_collision"]/surface/friction/ode/min_depth:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1467]: XML Element[min_depth], child of element[ode], not defined in SDF. Copying[min_depth] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="right_rear_wheel_link"]/collision[@name="right_rear_wheel_link_collision"]/surface/friction/ode/max_vel:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1549]: XML Element[max_vel], child of element[ode], not defined in SDF. Copying[max_vel] as children of [ode].
Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="right_rear_wheel_link"]/collision[@name="right_rear_wheel_link_collision"]/surface/friction/ode/min_depth:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1550]: XML Element[min_depth], child of element[ode], not defined in SDF. Copying[min_depth] as children of [ode].
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.
(2024-08-30 17:21:46.629) [error] Error Code 9: Msg: Failed to load a world.
```
### Issues
 - max_vel and min_depth inside ode tags from friction.
 - Uri can't find the model

Combuster54, 2024-08-30:
By looking documentation provided: http://sdformat.org/spec?elem=collision&ver=1.6

ode tag doesn't have max_vel and min_depth tags. So the solution is remove those tags from tunnel.sdf

Now, I will look into the other issue

Combuster54, 2024-09-05:
## Update

Hi. I was able to solve those issues. 

**First Issue**

`Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_front_wheel_link"]/collision[@name="left_front_wheel_link_collision"]/surface/friction/ode/max_vel:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1300]: XML Element[max_vel], child of element[ode], not defined in SDF. Copying[max_vel] as children of [ode].Warning [Utils.cc:132] [/sdf/world[@name="diff_drive"]/model[@name="vehicle"]/link[@name="left_rear_wheel_link"]/collision[@name="left_rear_wheel_link_collision"]/surface/friction/ode/min_depth:/usr/share/gz/gz-sim9/worlds/tunnel.sdf:L1384]: XML Element[min_depth], child of element[ode], not defined in SDF. Copying[min_depth] as children of [ode].
`

By looking documentation in SDF. `max_vel` and `min_depth` are not supported in ode tag.

**Solution** 
remove those tags

 
**Second Issue**

`(2024-08-30 17:21:46.629) [error] Error Code 14: Msg: Parser configurations requested resolved uris, but uri [tunnel_tile_1/materials/scripts] could not be resolved.`

**Solution**

Comparing other Tunnels models in Gazebo Fuel. I can find inside Tunnel Tile 1 some wrong paths inside `visual name="visual_ground"` inside script tag

```xml
    <!-- Tunnel Tile 1/model.sdf-->
   <visual name="visual_ground">
        <material>
          <script>
              <uri>tunnel_tile_1/materials/scripts</uri>
              <uri>tunnel_tile_1/materials/textures</uri>
              <name>TunnelTile/Ground_Diffuse</name>
          </script>
        </material>
      </visual>
```

```xml
    <!-- Tunnel Tile 1/model.sdf-->
   <visual name="visual_ground">
        <material>
          <script>
              <uri>materials/scripts</uri>
              <uri>materials/textures</uri>
              <name>TunnelTile/Ground_Diffuse</name>
          </script>
        </material>
      </visual>
```
## Result

[image]

Now I'm facing these new errors. But Even with could resolve these files. I was able to see the texture in gazebo
`
(2024-09-04 20:26:28.369) [error] [GUI] Could not resolve file [Phone.png]
(2024-09-04 20:26:28.433) [error] [GUI] Could not resolve file [Toolbox.png]
(2024-09-04 20:26:28.471) [error] [GUI] Could not resolve file [Extinguisher.png]`

`(2024-09-04 20:26:28.780) [error] [GUI] Triangle input has a count of zero. This is likely not desired`

It's relate to meshes

iche033, 2024-09-06:
thanks for finding the issue of the crash. I've updated Tunnel Tile 1 on Fuel to fix the script URIs.

You can remove the cache (`rm -r ~/.gz/fuel/fuel.gazebosim.org/openrobotics/models/tunnel\ tile\ 1/`) and run the world again to force gz sim to redownload the model. It should no longer crash. 

> `(2024-09-04 20:26:28.780) [error] [GUI] Triangle input has a count of zero. This is likely not desired`

This msg should no longer appear once https://github.com/gazebosim/gz-common/pull/635 is merged. 

> (2024-09-04 20:26:28.369) [error] [GUI] Could not resolve file [Phone.png]

The mesh dae file is unable to find the texture file but it should be fine as we override the material in SDF. So it's safe to ignore but the msg can be misleading.

Combuster54, 2024-09-06:
## Updated 

**Removing Tunnel Tile 1**

Now there are no errors related to Tunnel Tile 1. 

[image]

by reading @iche033 comments I assume that now tunnel.sdf is fine and works as expected, right?

utkuesen, 2024-09-06:
I tested it and get those errors.

$ gz sim /usr/share/gz/gz-sim9/worlds/tunnel.sdf
(2024-09-06 19:48:49.483) [error] Error in REST request

libcurl: (56) Recv failure: Connection reset by peer
(2024-09-06 19:48:49.483) [error] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openrobotics/models/subt_tunnel_staging_area/tip/subt_tunnel_staging_area.zip
  REST response code: 0
(2024-09-06 19:48:56.669) [error] Error in REST request

libcurl: (56) Recv failure: Connection reset by peer
(2024-09-06 19:48:56.669) [error] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: chapulina/models/extinguisher pbr/tip/extinguisher pbr.zip
  REST response code: 0
(2024-09-06 19:49:03.934) [error] Error in REST request

libcurl: (56) Recv failure: Connection reset by peer
(2024-09-06 19:49:03.935) [error] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openrobotics/models/tunnel tile 2/tip/tunnel tile 2.zip
  REST response code: 0
(2024-09-06 19:49:11.117) [error] Error in REST request

libcurl: (56) Recv failure: Connection reset by peer
(2024-09-06 19:49:11.118) [error] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openrobotics/models/tunnel tile 5/tip/tunnel tile 5.zip
  REST response code: 0

Then I removed "max_vel" and "min_depth" properties from ode tag. However, it didn't change.
I think there is a model download error because my computer has new Ubuntu 24.04.1 and there is no model folder created under "~/.gz/fuel/fuel.gazebosim.org/openrobotics/" directory, even i run the example.

Combuster54, 2024-09-06:
Uhmmm... Let me reinstall Gazebo Ionic to see if I can get the same error or similar

Combuster54, 2024-09-06:
```bash
combuster@combuster54:~$ gz sim /home/combuster/gazebo_issues/gz-sim/examples/worlds/tunnel.sdf 
(2024-09-06 10:00:42.444) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.444) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.444) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.444) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.446) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.446) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.446) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.446) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.447) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.448) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.448) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.448) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.450) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.450) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.451) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.452) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.475) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.489) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.490) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.543) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.543) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.543) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.543) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.546) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.546) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.546) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.546) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.547) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.547) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.547) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.547) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.550) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.550) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.552) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.552) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.571) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.571) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.572) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.573) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.576) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.576) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.576) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.576) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.578) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.578) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.578) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.579) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.625) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.656) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.658) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.664) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.664) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.665) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.667) [error] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:42.682) [error] Could not resolve file [Phone.png]
(2024-09-06 10:00:42.684) [error] Could not resolve file [Toolbox.png]
[... cut ...]
(2024-09-06 10:00:54.372) [error] [GUI] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:54.373) [error] [GUI] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:54.373) [error] [GUI] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:54.373) [error] [GUI] Triangle input has a count of zero. This is likely not desired
(2024-09-06 10:00:54.373) [error] [GUI] Triangle input has a count of zero. This is likely not desired
```

I find the same issues as we talk about before

iche033, 2024-09-06:
> I think there is a model download error because my computer has new Ubuntu 24.04.1 and there is no model folder created under "~/.gz/fuel/fuel.gazebosim.org/openrobotics/" directory, even i run the example.

gazebo should create the directory for you if it does not exist. From your error msgs (`(2024-09-06 19:48:56.669) [error] Error in REST request`), I think there's an issue with curl downloading models from Fuel. Hmm I'm not able to reproduce this yet as well.

utkuesen, 2024-09-07:
Hi, 
I am normally working behind proxy at work. Therefore, I tried the example with my phone's internet in order to be free at first time. Then, I got the errors which i showed my previous comment.
I tried it again with my internet at home today and it worked.
There are some errors about GUI which is shown by @Combuster54 . However, I could travel inside the tunnel with keyboard keys.
The problem I got early can be about internet issues.
I'm sorry, if I've bothered you for no reason.

[image]

iche033, 2024-09-09:
great thanks for confirming that it's working for you now.

closed 2024-09-09 by iche033