gazebosim/gazebo_test_cases #1691: gz-sim: velocity_control.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/velocity_control.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

andreasBihlmaier, 2024-08-29:
Will test

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 14-27-11.webm](https://github.com/user-attachments/assets/93a2088d-e60c-4971-8b95-47fd2ca1da4b)
LGTM

utkuesen, 2024-09-05:
I tested the example and it worked.
I recommend adding more instructions in order to easly run example by newbies, like me :)
Instructions can be change like that

1) Open Gazebo from terminal
gz sim

2) Search "velocity_control" from Search bar at the right side of screen. Then select "velocity_control.sdf" from list and click on "Run".

3) After Gazebo is opened, click on "Play" button at the left_bottom side of the screen.

4) Open a new terminal and run commands below.
gz topic -t "/model/vehicle_blue/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.05}"

gz topic -t "/model/vehicle_green/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 1.0}, angular: {z: -0.1}"

5) Change parameters of command and rerun.
gz topic -t "/model/vehicle_blue/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 0.0}, angular: {z: 0.5}"

gz topic -t "/model/vehicle_green/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 0.0}, angular: {z: -1.0}"

ahcorde, 2024-09-06:
Thank you for testing!

closed 2024-09-06 by ahcorde