gazebosim/gazebo_test_cases #1698: gz-sim: world_joint.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/world_joint.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-29:
Interested

Combuster54, 2024-08-30:
# Ubuntu 20.04 Noble

Starting world_joint.sdf 

```
gz sim world_joint.sdf
```
[image]

The joint is already attached to the world. Also, gz topic command shows data that validates its movement.
```
gz topic -t /world/shapes/dynamic_pose/info -e
```
[image]

Let me know if this the correct fortmat. In order to contribute in more issues

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde