gazebosim/gazebo_test_cases #1703: docs: install_osx

azeey-test, 2024-08-29:
## Configuration

* os: MacOS
* buildType: Binary

## Links

* [Live](https://gazebosim.org/docs/ionic/install_osx)
* [Source](https://github.com/gazebosim/docs/tree/master/ionic/install_osx.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

JeremyDWilliams, 2024-08-30:
I'd like to take this one

j-rivero, 2024-09-06:
Thanks so much for testing and welcome to the Gazebo community.

> The default value for the command "ulimit -n" is set to 256 which is too low. Running "ulimit -n 10240" and then proceeding with the homebrew install of gazebo worked! It might be useful to add this information to troubleshooting under macOS.

Could you please create a PR with this?

JeremyDWilliams, 2024-09-06:
According to the documentation Sonoma is not supported. I will do installs on Maverick and Ventura, but thought I'd try with Sonoma first.

After installing Homebrew from their page, the following install commands were entered in terminal:

brew tap osrf/simulation
brew install gz-ionic

This was the result:

> ==> Tapping osrf/simulation
> Cloning into '/opt/homebrew/Library/Taps/osrf/homebrew-simulation'...
> fatal: unable to access 'https://github.com/osrf/homebrew-simulation/': Could not resolve host: github.com
> Error: Failure while executing; `git clone https://github.com/osrf/homebrew-simulation /opt/homebrew/Library/Taps/osrf/homebrew-simulation --origin=origin --template= --config core.fsmonitor=false` exited with 128.
> ==> Downloading https://formulae.brew.sh/api/formula.jws.json
> curl: (6) Could not resolve host: formulae.brew.sh
> 
> Warning: formula.jws.json: update failed, falling back to cached version.
> ==> Downloading https://formulae.brew.sh/api/cask.jws.json
> curl: (6) Could not resolve host: formulae.brew.sh
> 
> Warning: cask.jws.json: update failed, falling back to cached version.
> Warning: No available formula with the name "gz-ionic".
> ==> Searching for similarly named formulae and casks...
> Error: No formulae or casks found for gz-ionic.

This was solved by setting PATH: as suggested by Homebrew, however it may not be immediately clear to the user.

I recommend adding verbiage to remind the user that PATH needs to be set for Homebrew to find what it needs.

JeremyDWilliams, 2024-09-06:
I suggest something like this:

**Binary Installation on MacOS**

All the Ionic binaries are available in Monterey and Ventura using the [([Homebrew package manager](https://brew.sh/))]. 

After installing Homebrew and adding it to your PATH: (directions in Homebrew's "Next steps:" install dialogue) Gazebo Ionic can be installed running:

azeey, 2024-09-06:
> fatal: unable to access 'https://github.com/osrf/homebrew-simulation/': Could not resolve host: github.com

The error message looks like there was a network issue and not an issue related to `PATH`. Can you confirm?

Also, we had more detailed instructions before, but it seemed redundant to have the homebrew install instructions here (see https://github.com/gazebosim/docs/pull/414)

JeremyDWilliams, 2024-09-08:
I'll do some uninstalls and fresh install to see if I can duplicate it. 

I did try twice before and had the same results until setting PATH. But it could have been a coincidence. Will test some more 😊

Some problems with GUI after install btw, but will deal with that elsewhere another time as I know GUI issues are not a new thing for Mac's.

scpeters, 2024-09-09:
> Thanks so much for testing and welcome to the Gazebo community.
>
>
> Could you please create a PR with this?

added in https://github.com/gazebosim/docs/pull/505

closed 2024-09-10 by azeey

JeremyDWilliams, 2024-09-16:
Tested using Venture and Maverick. No problems.