gazebosim/gazebo_test_cases #1707: [QT] Failed to create OpenGL context for format QSurfaceFormat

Roboticistprogrammer, 2024-09-06:
## Environment
* OS Version: Ubuntu ionic
* Source or binary build? Source Installation on Docker
 

## Description
* Expected behavior: I have successfully build the packages (although I have to admit it requires huge RAM, otherwise it doesn't work even if use --parallel-works 1 flag). I sourced my environment and run "gz sim shapes.sdf" to start gazebo.
* Actual behavior: I get: [QT] Failed to create OpenGL context for format QSurfaceFormat

## Steps to reproduce

1.Build the Pakcage.
2.Source the environment
3.gz sim shapes.sdf

## Output
[image]

Roboticistprogrammer, 2024-09-06:
Couple of Updates:
Apparently docker can not access X11 running on the host machine:
[image]

As you can see I have X11 running on the host machine, I authorized connection but Docker can not connect, thus I get can't display error. any suggestion?

Roboticistprogrammer, 2024-09-06:
**Note**:
I have tried running my container with rocker, also use -v flag as follows: 
[image]
 
and I got:
[image]

But still gazebo doesn't launch:

[image]

iche033, 2024-09-06:
can you try `gz sim -v 4 --headless-rendering shapes.sdf` and see if it makes any difference?

Roboticistprogrammer, 2024-09-08:
@iche033 Hi there, unfortunately I couldn't save my container while executing them with rocker, I am trying to have the most impact on this repo so I will continue on WSL, I will hop back on Docker after 13th and continue from here, just wanted to mention so it is not closed.

azeey, 2024-09-10:
I'm going to transfer this issue to gz-sim since this repo is specifically for tutorial test cases, I.e, each issue should be about testing a tutorial or running a manual test of a Gazebo functionality. Specific problems with running `gz sim` should be discussed in a gz-sim issue.