gazebosim/gazebo_test_cases #1710: docs: getstarted

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/jetty/getstarted)
* [Source](https://github.com/gazebosim/docs/tree/master/get_started.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-27:
I would like to take this one

iche033, 2025-08-27:
would you like to create a PR to update the doc?

ahcorde, 2025-08-28:
Do you mind to update the docs ? This is the [docs repository](https://github.com/gazebosim/docs)

And this is the file

https://github.com/gazebosim/docs/blob/master/get_started.md