gazebosim/gazebo_test_cases #1734: docs: fuel_insert

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/jetty/fuel_insert)
* [Source](https://github.com/gazebosim/docs/tree/master/jetty/Model_insertion_fuel.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

akky20, 2025-08-30:
[image]

Working fine, no issues !!
[status: passed]

ahcorde, 2025-09-01:
Thank you for testing!