gazebosim/gazebo_test_cases #1737: docs: hotkeys

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/jetty/hotkeys)
* [Source](https://github.com/gazebosim/docs/tree/master/jetty/hotkeys.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

s0um0r0y, 2025-08-27:
# Brief

- I have tested pause and unpause and snap while rotating and translating the hotkeys for them work, there is no screenshot that I can attach for that
- below is the image for saving with `ctrl+s` command

[image]

- below is the image for saving with `ctrl+shift+s` command, but in this case I am not able to save the `.config` file
as I am not able to see a `save` button,  I can only see the `open` button on the GUI
[image]

- search menu shortcuts works below
[image]

- below is the image for opening a saved configuration using `ctrl+o` to open `gui.config` and that does not seem to work attaching the screenshot below

[image]

 
-  shut down command `ctrl+q` works

`[status: failed]`

Physic69, 2025-08-27:
[ source installed, Noble 24.04]
Few hotkeys are giving trouble, like ctrl+shit+s is not saving the file its supposed to after pressing save at desired location

Space bar is working fine, Ctrl+ drag is also working, Shift is also working well!
(well i cant show those in a pic, just putting one which is taken after playing around with those)

[image]

Ctrl+S:
[image]

Ctrl+Shift+S: (unable to save)
[image]

/ :(search menu)
[image]

Ctrl+q: (closing properly) 
(Cant be showed ig)

Ctrl+o: (The config file is supposed to stop the pendulum (freeze), with real time factor as 0, but it didn't work as expected ) 
[image]

[status: failed]

closed 2025-09-16 by caguero