gazebosim/gazebo_test_cases #1740: docs: spawn_urdf

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/docs/jetty/spawn_urdf)
* [Source](https://github.com/gazebosim/docs/tree/master/jetty/spawn_urdf.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-29:
Robot spawns correctly and works as expected. There are error messages in the gz sim terminal, but I assume they are related to missing ROS packages.

[image]

[status: passed]

iche033, 2025-08-30:
Thanks for testing. Yes the errors in red are fine, I created #616 to get rid of those errors.