gazebosim/gazebo_test_cases #1741: docs: spawn_urdf

azeey-test, 2025-08-27:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/docs/jetty/spawn_urdf)
* [Source](https://github.com/gazebosim/docs/tree/master/jetty/spawn_urdf.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Creator-1705, 2025-09-09:
[image]
[image]
[image]

[status: passed]

ahcorde, 2025-09-10:
Thank you for testing!