gazebosim/gazebo_test_cases #1749: gz-common: Profiler

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/common/7/profiler.html)
* [Source](https://github.com/gazebosim/gz-common/tree/gz-common7/tutorials/profiler.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

chen-harrison, 2025-09-03:
[image]

Platform: Ubuntu Noble 24.04 via Docker container (ghcr.io/gazebosim/gz-jetty)

Got it to work, but the documentation had some gaps and there was a good amount of trial and error involved.

### Possible Documentation Fixes

- When it says to download the profiler.cc example, it should tell you that this should go into into a new project directory, e.g. "Run `mkdir -p ~/profiler_example && cd ~/profiler_example`". This is because it ends up being its own CMake project.
- Just capture profile.cc via a `wget` command instead of a link that opens to the raw text: `wget https://raw.githubusercontent.com/gazebosim/gz-common/gz-common7/examples/profiler.cc`
- "Update your CMakeLists.txt" should be "create a new CMakeLists.txt in the ~/profiler_example directory" - it doesn't exist yet and I wasn't sure if I was supposed to have an existing file
- Needs a `mkdir build` before `cd build` - again, it doesn't exist yet (unless my understanding is wrong)
- The commands for terminal 2 should be separated out, and the explanations being comments make the code block seem like one large script. I think it's better to have those explanations outside of the code block
- The `python3 -m http.server $SOURCE_DIR/gz-common/profiler/src/Remotery/vis/index.html` errors out and seems outdated. Instead, in my Docker container (provided by `gazebosim` container registry) I ran `cd /usr/share/gz/gz-common/profiler_vis` to get to the directory with `index.html`, then ran `python3 -m http.server`. When I did inside the container, it published on http://0.0.0.0:8000 instead, though I'm not sure it's related to how I'm launching the container

I can create a PR depending on if these comments are sound/make sense.

[status: passed]

azeey, 2025-09-03:
Those are all really good points. On the last point about running `python3 -m http.server`, I believe the `-d` flag must be used and the directory given instead of the `index.html` (see https://docs.python.org/3/library/http.server.html#command-line-interface), so it should have been `python3 -m http.server -d $SOURCE_DIR/gz-common/profiler/src/Remotery/vis`. This of course assumes `$SOURCE_DIR` is defined, but I think we can be more explicit about it.

> When I did inside the container, it published on http://0.0.0.0:8000/ instead, though I'm not sure it's related to how I'm launching the container

You'll have to expose port 8000. https://docs.docker.com/get-started/docker-concepts/running-containers/publishing-ports/#publishing-ports

With all that being said, please do create a PR with all of your suggestions. And thank you for testing!

closed 2025-09-03 by azeey