gazebosim/gazebo_test_cases #1752: gz-fuel-tools: Installation

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/fuel_tools/11/install.html)
* [Source](https://github.com/gazebosim/gz-fuel-tools/tree/gz-fuel-tools11/tutorials/01_installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Creator-1705, 2025-08-30:
[image]

was able to install it

caguero, 2025-09-01:
Thanks for checking!