gazebosim/gazebo_test_cases #1761: gz-fuel-tools: C++ API

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/fuel_tools/11/cppapi.html)
* [Source](https://github.com/gazebosim/gz-fuel-tools/tree/gz-fuel-tools11/tutorials/04_cpp_api.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Creator-1705, 2025-08-30:
[image]

unable to get the output of :
./list -t world

[status: failed]

caguero, 2025-09-03:
I don't see that the `make` command actually compile any files. Could you double check your steps?

CursedRock17, 2025-09-05:
So, I recreated this and everything went smoothly, docs were clear as well.
**Operating System:** Ubuntu 24.04
**Source/Binary:** Source

[image]

[image]

[status: passed]

ahcorde, 2025-09-05:
Thank you for testing