gazebosim/gazebo_test_cases #1779: gz-sim: Blender Python script for programmatic mesh distortion

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/blender_distort_meshes.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/blender_distort_meshes.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

SuperGops7, 2025-09-04:
[ source installed, Noble 24.04]

The tutorial makes it clear on how to run the distortion scripts without any issues. 
The script works fine in distorting the target object.
[image]

[status: passed]

ahcorde, 2025-09-04:
Thank you for testing!