gazebosim/gazebo_test_cases #1845: gz-sim: Light config

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/light_config.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/light_config.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

s0um0r0y, 2025-08-29:
# Brief 

- This is working and I have verified the same

[image]

`[status: passed]`

nikodemj9, 2025-08-29:
[image]

[status: passed]

closed 2025-08-29 by iche033