gazebosim/gazebo_test_cases #1849: gz-sim: Logging

azeey-test, 2025-08-27:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/10/log.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/log.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Creator-1705, 2025-09-09:
[image]

[status: passed]

mukul2020, 2025-09-09:
I'm not able to find any grammatical mistake 

[status: passed]

closed 2025-09-16 by bperseghetti