gazebosim/gazebo_test_cases #1851: gz-sim: Logical Audio Sensor

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/logicalaudiosensor.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/logical_audio_sensor.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

akky20, 2025-08-28:
[image]

### No output (as desired)
[image]

[image]

Seems good to me , found no typos and all hyperlinks works !
[status: passed]

ahcorde, 2025-08-29:
Thank you for testing!

closed 2025-08-29 by ahcorde