gazebosim/gazebo_test_cases #1860: gz-sim: Model Command

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/model_command.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/model_command.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

ahcorde, 2025-08-29:
Thank you for testing!

closed 2025-08-29 by ahcorde

AronLapp, 2025-09-09:
Setup
- Container: ghcr.io/gazebosim/gz-jetty:main
- OS (inside container): Ubuntu 24.04 (Noble)
- Test: gz model command with the diff_drive.sdf world

Steps:

1. Started simulation with gz sim -r diff_drive.sdf.
2. Verified available models with gz model --list.
3. Queried full model info for vehicle_blue.
4. Retrieved pose with --pose.
5. Retrieved a single link with --link caster.
6. Retrieved a single joint with --joint caster_wheel.

```bash
root@e02caca9697d:/# gz model --list

Requesting state for world [diff_drive]...

Available models:
    - ground_plane
    - vehicle_blue
    - vehicle_green
root@e02caca9697d:/# gz model -m vehicle_blue

Requesting state for world [diff_drive]...

Model: [8]
  - Name: vehicle_blue
  - Pose [ XYZ (m) ] [ RPY (rad) ]:
    [-0.000000 2.000000 0.324999]
    [-0.000000 -0.000001 -0.000000]
  - Link [9]
    - Name: chassis
    - Parent: vehicle_blue [8]
    - Mass (kg): 1.143950
    - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Inertial Matrix (kg.m^2):
      [0.126164 0.000000 0.000000]
      [0.000000 0.416519 0.000000]
      [0.000000 0.000000 0.481014]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [-0.151427 0.000000 0.175000]
      [0.000000 -0.000000 0.000000]
  - Link [12]
    - Name: left_wheel
    - Parent: vehicle_blue [8]
    - Mass (kg): 2.000000
    - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Inertial Matrix (kg.m^2):
      [0.145833 0.000000 0.000000]
      [0.000000 0.145833 0.000000]
      [0.000000 0.000000 0.125000]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.554283 0.625029 -0.025000]
      [-1.570700 -0.000000 -0.000000]
  - Link [15]
    - Name: right_wheel
    - Parent: vehicle_blue [8]
    - Mass (kg): 2.000000
    - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Inertial Matrix (kg.m^2):
      [0.145833 0.000000 0.000000]
      [0.000000 0.145833 0.000000]
      [0.000000 0.000000 0.125000]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.554282 -0.625029 -0.025000]
      [-1.570700 -0.000000 -0.000000]
  - Link [18]
    - Name: caster
    - Parent: vehicle_blue [8]
    - Mass (kg): 1.000000
    - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Inertial Matrix (kg.m^2):
      [0.100000 0.000000 0.000000]
      [0.000000 0.100000 0.000000]
      [0.000000 0.000000 0.100000]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [-0.957138 0.000000 -0.124999]
      [-0.000000 0.000006 -0.000000]
  - Joint [21]
    - Name: left_wheel_joint
    - Parent: vehicle_blue [8]
    - Type: revolute
    - Parent Link: chassis [9]
    - Child Link: left_wheel [12]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Axis unit vector [ XYZ ]:
      [0 0 1]
  - Joint [22]
    - Name: right_wheel_joint
    - Parent: vehicle_blue [8]
    - Type: revolute
    - Parent Link: chassis [9]
    - Child Link: right_wheel [15]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [-0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
    - Axis unit vector [ XYZ ]:
      [0 0 1]
  - Joint [23]
    - Name: caster_wheel
    - Parent: vehicle_blue [8]
    - Type: ball
    - Parent Link: chassis [9]
    - Child Link: caster [18]
    - Pose [ XYZ (m) ] [ RPY (rad) ]:
      [0.000000 0.000000 0.000000]
      [0.000000 -0.000000 0.000000]
root@e02caca9697d:/# gz model -m vehicle_blue --pose

Requesting state for world [diff_drive]...

Model: [8]
  - Name: vehicle_blue
  - Pose [ XYZ (m) ] [ RPY (rad) ]:
    [-0.000000 2.000000 0.324999]
    [-0.000000 -0.000001 -0.000000]
root@e02caca9697d:/# gz model -m vehicle_blue --link caster

Requesting state for world [diff_drive]...

- Link [18]
  - Name: caster
  - Parent: vehicle_blue [8]
  - Mass (kg): 1.000000
  - Inertial Pose [ XYZ (m) ] [ RPY (rad) ]:
    [0.000000 0.000000 0.000000]
    [0.000000 -0.000000 0.000000]
  - Inertial Matrix (kg.m^2):
    [0.100000 0.000000 0.000000]
    [0.000000 0.100000 0.000000]
    [0.000000 0.000000 0.100000]
  - Pose [ XYZ (m) ] [ RPY (rad) ]:
    [-0.957138 0.000000 -0.124999]
    [-0.000000 0.000006 -0.000000]
root@e02caca9697d:/# gz model -m vehicle_blue --joint caster_wheel

Requesting state for world [diff_drive]...

- Joint [23]
  - Name: caster_wheel
  - Parent: vehicle_blue [8]
  - Type: ball
  - Parent Link: chassis [9]
  - Child Link: caster [18]
  - Pose [ XYZ (m) ] [ RPY (rad) ]:
    [0.000000 0.000000 0.000000]
    [0.000000 -0.000000 0.000000]
root@e02caca9697d:/# 
```

[image]

[status: passed]