gazebosim/gazebo_test_cases #1873: gz-sim: Particle Emitter

azeey-test, 2025-08-27:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/10/particle_emitter.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/particle_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Creator-1705, 2025-09-09:
[image]

[status: passed]

caguero, 2025-09-09:
Thanks for checking! Shouldn't we visualize smoke?

ahcorde, 2025-09-10:
Thank you for testing!

azeey, 2025-09-10:
@mukul2020 Just wanted to confirm you tested this on windows. Were you able to build Gazebo from source?

closed 2025-09-16 by bperseghetti