gazebosim/gazebo_test_cases #1875: gz-sim: Pause and Run simulation

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/pause_run_simulation.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/pause_run_simulation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

matosinho, 2025-08-27:
Tested on binary installation. The pause/play command was tested from command line and from the GUI. The stats was monitored and showed correctly `pause: true` whenever the simulation was paused. If the simulation was not, the `pause` attribute did not appear, which is the default behaviour explained in the tutorial.

Here the `gif` showing the tests:

[image]

`[status: passed]`

ahcorde, 2025-08-28:
thank you for testing!

closed 2025-08-28 by ahcorde