gazebosim/gazebo_test_cases #1922: gz-sim: Triggered Publisher

azeey-test, 2025-08-27:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/sim/10/triggeredpublisher.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/triggered_publisher.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

jasmeet0915, 2025-09-05:
The example world works as expected:

[image]

[status: passed]

iche033, 2025-09-05:
thanks for testing!