gazebosim/gazebo_test_cases #1933: gz-sim: Video Recorder

azeey-test, 2025-08-27:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/10/videorecorder.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/video_recorder.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

azeey, 2025-09-05:
Hi @Creator-1705 I'm not sure if you saw the section in the readme about testing on Windows (https://github.com/gazebosim/gazebo_test_cases?tab=readme-ov-file#windows-tests). Were you able to build all the packages from source?

Creator-1705, 2025-09-05:
> Hi [@Creator-1705](https://github.com/Creator-1705) I'm not sure if you saw the section in the readme about testing on Windows (https://github.com/gazebosim/gazebo_test_cases?tab=readme-ov-file#windows-tests). Were you able to build all the packages from source?

no,
is that the reason for not working?

closed 2025-09-16 by bperseghetti