gazebosim/gazebo_test_cases #1935: gz-sim: Websocket server for gzweb

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/10/websocket_server.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/tutorials/websocket_server.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-29:
Gazebo is starting in server mode, websocket server seems to be initalized correctly after calling gz service, scene is visible from https://app.gazebosim.org/visualization connected to localhost.

[image]

[status: passed]

iche033, 2025-08-29:
nice, thanks for testing

closed 2025-08-29 by iche033