gazebosim/gazebo_test_cases #1938: gz-gui: Installation

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/gui/10/install.html)
* [Source](https://github.com/gazebosim/gz-gui/tree/gz-gui10/tutorials/01_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

XINJIANGMO, 2025-09-01:
[image]
[image]
[status: failed]

AronLapp, 2025-09-07:
Environment:
- Jetty container: ghcr.io/gazebosim/gz-jetty:main (Ubuntu 24.04 Noble)
-  Host: Ubuntu 22.04, Intel Iris (X11 passthrough only for build; tests run headless)
- OSRF repos pre-installed in container

Steps

```bash
git clone https://github.com/gazebosim/gz-gui
cd gz-gui
apt-get update
apt-get install -y $(sort -u $(find . -iname 'packages.apt') | tr '\n' ' ')

mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=ON ..
make -j"$(nproc)"

xvfb-run -s "-screen 0 1280x800x24" \
  ctest -j"$(nproc)" --output-on-failure | tee /tmp/ctest_full_xvfb.log

make codecheck | tee /tmp/codecheck.log
```

Results:
- Build: Success
- CTest: 55/56 tests passed, 1 failed
  - Failure: UNIT_Publisher_TEST
  - Note: without xvfb-run additional Qt-related tests fail (UNIT_gz_TEST, UNIT_TopicEcho_TEST) becuase Qt cant create a display in headless mode.
- Codecheck finished clean
  - "include file not found" errors appear when running cppcheck manually without project config
  - project's make codecheck target calls cppcheck with the correct config, so no relevant errors appear

log tails:
```
ctest (xvfb): 98% tests passed, 1 tests failed out of 56
  The following tests FAILED:
    29 - UNIT_Publisher_TEST (Failed)
```
```
codecheck:
Built target cpplint
Built target cppcheck
Built target codecheck
```

Outcome:
- Source install works
- Tests: 55/56 (1 fail using Xvfb)
- Static analysis clean

[status: failed]

caguero, 2025-09-16:
Thanks for checking!

closed 2025-09-16 by caguero