gazebosim/gazebo_test_cases #1971: gz-launch: Gazebo launch tutorial

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/launch/9/basics.html)
* [Source](https://github.com/gazebosim/gz-launch/tree/gz-launch9/tutorials/tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

s0um0r0y, 2025-08-29:
I think this is the command to install `gz-launch` for this task right

```
sudo apt install libgz-launch9
```

it says that this is depreciated in this page https://github.com/gazebosim/gz-launch/tree/gz-launch9?tab=readme-ov-file#install

Gazebo Launch is deprecated in Gazebo Jetty (gz-launch9) and will be removed in Gazebo K. Please see the [migration guide](https://github.com/gazebosim/gz-launch/blob/main/Migration.md).

s0um0r0y, 2025-08-29:
I have done some analysis and I have realised that the issue is here in the factory.gzlaunch its in the sim factory plugin

`<plugin name="gz::launch::SimFactory"`

iche033, 2025-08-30:
Thanks for testing. I made some fixes to the tutorial doc and example launch file in https://github.com/gazebosim/gz-launch/pull/309