gazebosim/gazebo_test_cases #1980: gz-math: Angle example

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/math/9/example_angle.html)
* [Source](https://github.com/gazebosim/gz-math/tree/gz-math9/tutorials/example_angle.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

pratik-adhikari, 2025-08-28:
[image]

`#include <iostream>
#include <gz/math/Angle.hh>
#include <gz/math/config.hh>

int main()
{
    std::cout << "Testing gz-math Angle Example" << std::endl;
    std::cout << "gz-math version: " << GZ_MATH_VERSION_FULL << std::endl;
    std::cout << "==============================" << std::endl;

    gz::math::Angle a;
    std::cout << "The angle 'a' should be zero: " << a << std::endl;

    a = gz::math::Angle::Pi;
    std::cout << "Pi in radians: " << a << std::endl;
    std::cout << "Pi in degrees: " << a.Degree() << std::endl;

    a += gz::math::Angle::HalfPi;
    std::cout << "pi + PI/2 in radians: " << a << std::endl;

    std::cout << "Normalized to the range -pi and Pi: " 
              << a.Normalized() << std::endl;

    return 0;
}
`
Passed the tutorial
[status: passed]

ahcorde, 2025-08-28:
Thank you for testing!