gazebosim/gazebo_test_cases #2000: gz-msgs: Installation

azeey-test, 2025-08-27:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/msgs/12/install.html)
* [Source](https://github.com/gazebosim/gz-msgs/tree/gz-msgs12/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

saiaravind19, 2025-09-06:
Needs a doc update with the branch name. from ign-msgs<version> to gz-msgs<version>

I cloned the repo using ``` git clone https://github.com/gazebosim/gz-msgs -b gz-msgs12 ```

Build type: Source.
Build is successful without any issue. But when I run make test it shows that one of the test case is failed.

[image]

[status: failed]

caguero, 2025-09-16:
Thanks for testing!