gazebosim/gazebo_test_cases #2022: gz-physics: Implement physics feature

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/physics/9/createphysicsplugin.html)
* [Source](https://github.com/gazebosim/gz-physics/tree/gz-physics9/tutorials/07-implementing-a-physics-plugin.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

SuperGops7, 2025-08-28:
[Noble 24.04]

The tutorial is detailed and easy to follow.

[image]

The last command seems wrong, considering the name of the folder we are creating is hello_world_plugin, and not simple_plugin. The correct command should be:

`./hello_world_loader/build/hello_world_loader hello_world_plugin/build/libHelloWorldPlugin.so`

[status: failed]

SuperGops7, 2025-08-29:
Hey @ahcorde , you can find the PR [here](https://github.com/gazebosim/gz-physics/pull/777)

scpeters, 2025-08-29:
https://github.com/gazebosim/gz-physics/pull/777 has been merged