gazebosim/gazebo_test_cases #2033: gz-rendering: Installation

azeey-test, 2025-08-27:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/rendering/10/installation.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering10/tutorials/02_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

saiaravind19, 2025-09-06:
I am working on this

saiaravind19, 2025-09-06:
I build the repo from source.
Branch : gz-rendering10

Build is successful without any issues.

Next I started running ```make test ``` and it fails in some test cases. Log and screenshots are attached below for reference.

[image]

Log file of the test
[LastTest.log](https://github.com/user-attachments/files/22186258/LastTest.log)

Then I ran ```GZ_ENGINE_TO_TEST=ogre2 GZ_ENGINE_BACKEND=gl3plus ./bin/UNIT_Camera_TEST ``` . Screen shot for the same is attached below

[image]

[status: failed]

iche033, 2025-09-09:
thanks for testing! Do you mind running this failing test and post the output?

```
GZ_ENGINE_TO_TEST=ogre2 GZ_ENGINE_BACKEND=metal ./bin/UNIT_RenderingIface_TEST
```

saiaravind19, 2025-09-09:
[image]

@iche033  here is the screen shot.