gazebosim/gazebo_test_cases #2034: gz-rendering: Installation

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/rendering/10/installation.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering10/tutorials/02_install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

iche033, 2025-09-02:
from the output, the test was interrupted? Are there any particular tests that are failing? The 2 red error messages are fine as it is intentionally testing invalid params.

AronLapp, 2025-09-07:
Environment:
-Jetty container: ghcr.io/gazebosim/gz-jetty:main (Ubuntu 24.04 Noble)
- Host: Ubuntu 22.04, Intel Iris (X11 passthrough for build + GUI; tests headless with Xvfb)
- OSRF repos preinstalled in container

Steps:
```bash
apt-get update
apt-get install -y \
  build-essential g++ cmake pkg-config git xvfb \
  libglew-dev freeglut3-dev libxmu-dev libxi-dev \
  libogre-next-dev \
  libgz-cmake5-dev libgz-math9-dev libgz-common7-dev libgz-plugin4-dev

git clone https://github.com/gazebosim/gz-rendering
cd gz-rendering
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release ..
make -j"$(nproc)"

xvfb-run -s "-screen 0 1280x800x24" \
  env GZ_ENGINE_TO_TEST=ogre2 GZ_ENGINE_BACKEND=gl3plus \
  ctest --output-on-failure -j"$(nproc)" | tee /tmp/ctest_full_xvfb.log

```

Log tail:
´´´
98% tests passed, 4 tests failed out of 257

The following tests FAILED:
     86 - REGRESSION_reload_engine_ogre_gl3plus (Failed)
     88 - REGRESSION_reload_engine_ogre2_gl3plus (Failed)
    214 - UNIT_RenderingIface_TEST_ogre_gl3plus (Failed)
    216 - UNIT_RenderingIface_TEST_ogre2_gl3plus (Failed)
´´´

Outcome:
- source install works
- tests mostly pass, 4 fail (see above)
- static analysis not run

[status: failed]

avanmalleghem, 2025-09-08:
> from the output, the test was interrupted? Are there any particular tests that are failing? The 2 red error messages are fine as it is intentionally testing invalid params.

there is an interruption because it is stuck and can't go to the next step