gazebosim/gazebo_test_cases #2057: gz-rendering: Render pass

azeey-test, 2025-08-27:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/rendering/10/render_pass.html)
* [Source](https://github.com/gazebosim/gz-rendering/tree/gz-rendering10/tutorials/17_render_pass_tutorial.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

saiaravind19, 2025-09-06:
Able to build and run it without issues.

https://github.com/user-attachments/assets/5d3377a0-b6ff-4639-a2ec-9f784f34bbe4

[status: passed]

ahcorde, 2025-09-08:
Thank you for testing!

closed 2025-09-08 by ahcorde