gazebosim/gazebo_test_cases #2091: gz-sensors: Custom sensors

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sensors/10/custom_sensors.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors10/tutorials/custom_sensors.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

ahcorde, 2025-08-28:
Thank you for testing!

closed 2025-08-28 by ahcorde

shreya-ramesh, 2025-08-31:
works fine!!

[status: passed]