gazebosim/gazebo_test_cases #2100: gz-sensors: Thermal Camera in Gazebo

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sensors/10/thermalcameraigngazebo.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors10/tutorials/thermal_camera.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

pratik-adhikari, 2025-08-28:
[image]

[status: passed]
Thermal camera world file loading properly. 
(visualised added RGB in that world file)  

root@pratik-AORUS-15X-ASF:/tmp/gazebo_wide_angle_test# gz topic -l | grep thermal
/thermal_camera
/thermal_camera_8bit/camera_info
/thermal_camera_8bit/image
/world/thermal_camera/clock
/world/thermal_camera/dynamic_pose/info
/world/thermal_camera/pose/info
/world/thermal_camera/scene/deletion
/world/thermal_camera/scene/info
/world/thermal_camera/state
/world/thermal_camera/stats
/world/thermal_camera/light_config
/world/thermal_camera/material_color

ahcorde, 2025-08-28:
Thank you for testing!

closed 2025-08-28 by ahcorde