gazebosim/gazebo_test_cases #2122: gz-transport: Contribute

azeey-test, 2025-08-27:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/transport/15/contribute.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport15/tutorials/21_contribute.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

iche033, 2025-08-27:
for this one, it should be straightforward, just make sure the link works correctly

ahcorde, 2025-09-09:
I created the PR https://github.com/gazebosim/docs/pull/624 with the fixes

Thank you for testing

closed 2025-09-09 by ahcorde