gazebosim/gazebo_test_cases #2135: gz-sim: acoustic_comms_propagation.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/acoustic_comms_propagation.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

Narashima1808, 2025-08-28:
[image]

Using Docker, Ubuntu Noble

[status: passed]

akky20, 2025-08-30:
[image]

###my system

[image]

[status: passed]

closed 2025-09-01 by ahcorde