gazebosim/gazebo_test_cases #2159: gz-sim: depth_camera_sensor.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/depth_camera_sensor.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-29:
[image]

[status: passed]

ahcorde, 2025-08-29:
Thank you for testing!

closed 2025-08-29 by ahcorde