gazebosim/gazebo_test_cases #2182: gz-sim: joint_controller.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/joint_controller.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

closed 2025-08-28 by iche033

shreya-ramesh, 2025-08-29:
@iche033 hey may I know why this case is not shown on my leader board?

shreya-ramesh, 2025-08-29:
@ahcorde thank you for assigning! but it still is not shown on the board

iche033, 2025-08-29:
> [@ahcorde](https://github.com/ahcorde) thank you for assigning! but it still is not shown on the board

Not sure if it's because the issue is assigned after it is closed? cc @bperseghetti

bperseghetti, 2025-08-29:
> @iche033 hey may I know why this case is not shown on my leader board?
> 

You need a space between status: and passed. It's intentionally very exact so as not to accidentally trigger otherwise.