gazebosim/gazebo_test_cases #2197: gz-sim: log_record_resources.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/log_record_resources.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-28:
[image]

[status: passed]

AronLapp, 2025-09-08:
Environment:
    Jetty container (Ubuntu Noble base), host Ubuntu 22.04, Intel Iris (X11)
    X11 passthrough with /dev/dri

Recording logs:
```bash
gz sim -v4 ~/worlds/log_record_resources.sdf --record
```

[image]

[image]

Playback appears to work fine but throws some errors. The recorded file is still playable and all textures are visible in the graphic window

[image]

Output:
```shell
root@34252fbd47eb:~/gz-sim/build# LOGDIR=$(dirname "$(ls -td ~/.gz/sim/log/*/state.tlog | head -n1)")
ls -lah "$LOGDIR/state.tlog"
-rw-r--r-- 1 root root 1.7M Sep  8 12:16 /root/.gz/sim/log/2025-09-08T12:16:02.236009451/state.tlog
root@34252fbd47eb:~/gz-sim/build# mkdir -p "$LOGDIR/root/.gz"
cp -a ~/.gz/fuel "$LOGDIR/root/.gz/"
root@34252fbd47eb:~/gz-sim/build# export GZ_SIM_RESOURCE_PATH="$LOGDIR/root:${GZ_SIM_RESOURCE_PATH}"
gz sim --playback "$LOGDIR" -v4
(2025-09-08 12:34:09.866) [debug] [gz.cc:238] Subscribing to [/gazebo/starting_world].
(2025-09-08 12:34:09.866) [debug] [gz.cc:240] Waiting for a world to be set from the GUI...
(2025-09-08 12:34:09.888) [info] [Gui.cc:332] Gazebo Sim GUI    v10.0.0~pre1
(2025-09-08 12:34:09.889) [debug] [Gui.cc:392] Waiting for subscribers to [/gazebo/starting_world]...
(2025-09-08 12:34:09.990) [info] [gz.cc:242] Received world [] from the GUI.
(2025-09-08 12:34:09.990) [debug] [gz.cc:244] Unsubscribing from [/gazebo/starting_world].
(2025-09-08 12:34:09.991) [info] [gz.cc:407] Gazebo Sim Server v10.0.0~pre1
(2025-09-08 12:34:09.991) [info] [gz.cc:459] Playing back states/root/.gz/sim/log/2025-09-08T12:16:02.236009451
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
(2025-09-08 12:34:10.001) [debug] [Application.cc:97] Initializing application.
(2025-09-08 12:34:10.001) [debug] [Application.cc:171] Qt using OpenGL graphics interface
(2025-09-08 12:34:10.003) [info] [ServerPrivate.cc:737] Loading default world.
(2025-09-08 12:34:10.003) [debug] [Application.cc:685] [GUI] Create main window
(2025-09-08 12:34:10.105) [debug] [PathManager.cc:70] [GUI] Requesting resource paths through [/gazebo/resource_paths/get]
(2025-09-08 12:34:10.217) [debug] [Gui.cc:476] [GUI] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
(2025-09-08 12:34:11.867) [info] [SystemManager.cc:54] Serving entity system service on [/entity/system/add]
(2025-09-08 12:34:11.869) [info] [SimulationRunner.cc:279] Serving world controls on [/world/default/control], [/world/default/control/state] and [/world/default/playback/control]
(2025-09-08 12:34:11.869) [info] [SimulationRunner.cc:305] Serving GUI information on [/world/default/gui/info]
(2025-09-08 12:34:11.869) [info] [SimulationRunner.cc:308] World [default] initialized with [] physics profile.
(2025-09-08 12:34:11.869) [info] [SimulationRunner.cc:315] Serving world SDF generation service on [/world/default/generate_world_sdf]
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:357] Serving world names on [/gazebo/worlds]
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:370] Resource path add service on [/gazebo/resource_paths/add].
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:383] Resource path get service on [/gazebo/resource_paths/get].
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:398] Resource path resolve service on [/gazebo/resource_paths/resolve].
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:412] Resource paths published on [/gazebo/resource_paths].
(2025-09-08 12:34:11.869) [info] [ServerPrivate.cc:425] Server control service on [/server_control].
(2025-09-08 12:34:11.970) [info] [SimulationRunner.cc:762] Found no publishers on /stats, adding root stats topic
(2025-09-08 12:34:11.971) [info] [SimulationRunner.cc:796] Found no publishers on /clock, adding root clock topic
(2025-09-08 12:34:11.971) [debug] [PathManager.cc:59] [GUI] Received resource paths.
(2025-09-08 12:34:11.972) [info] [LevelManager.cc:646] Loaded level [default]
(2025-09-08 12:34:11.971) [debug] [Gui.cc:534] [GUI] Requesting GUI from [/world/default/gui/info]...
(2025-09-08 12:34:11.980) [debug] [ServerConfig.cc:1042] Loading (2) plugins from file [/root/.gz/sim/10/playback_server.config]
(2025-09-08 12:34:11.980) [info] [SimulationRunner.cc:1697] No systems loaded from SDF, loading defaults
(2025-09-08 12:34:11.981) [debug] [GuiRunner.cc:149] [GUI] Requesting initial state from [/world/default/state]...
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:752] Create service on [/world/default/create_multiple] (async)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:754] Create service on [/world/default/create_multiple/blocking] (blocking)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:752] Remove service on [/world/default/remove] (async)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:754] Remove service on [/world/default/remove/blocking] (blocking)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:752] Pose service on [/world/default/set_pose] (async)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:754] Pose service on [/world/default/set_pose/blocking] (blocking)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:752] Pose service on [/world/default/set_pose_vector] (async)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:754] Pose service on [/world/default/set_pose_vector/blocking] (blocking)
(2025-09-08 12:34:11.988) [info] [UserCommands.cc:752] Light configuration service on [/world/default/light_config] (async)
[... cut ...]
(2025-09-08 12:34:12.703) [error] [SystemPaths.cc:355] [GUI] Unable to find file with URI [16://02.236009451/root/.gz/fuel/fuel.gazebosim.org/openrobotics/models/x2 config 1/19/meshes]
(2025-09-08 12:34:12.704) [error] [SystemPaths.cc:355] [GUI] Unable to find file with URI [//root/.gz/sim/log/2025-09-08T12://16:02.236009451/root/.gz/fuel/fuel.gazebosim.org/openrobotics/models/x2 config 1/19/meshes]
(2025-09-08 12:34:12.705) [debug] [EntityContextMenuPlugin.cc:79] [GUI] Entity context menu plugin is using camera [scene::Camera(65527)]
```

[status: passed]
(passed functionally, with errors that dont affect functionality)

closed 2025-09-09 by caguero