gazebosim/gazebo_test_cases #2238: gz-sim: spherical_coordinates.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/spherical_coordinates.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

akky20, 2025-08-27:
[image]

[image]

#### My System

[image]

[status: passed]

Physic69, 2025-08-28:
[ source installed, Noble 24.04]
Well, the file works fine,
sdf file worked with no issues, even comments worked well.
[image]

[image]

my system:
Distributor ID: Ubuntu
Description: Ubuntu 24.04.3 LTS
Release: 24.04
Codename: noble

[status: passed]

ahcorde, 2025-08-28:
Thank you for testing!

closed 2025-08-28 by ahcorde