gazebosim/gazebo_test_cases #2249: gz-sim: video_record_dbl_pendulum.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/video_record_dbl_pendulum.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

caguero, 2025-09-03:
Thanks for going though this tutorial! Out of curiosity, what's your OS? I can't reproduce it on Ubuntu.

AronLapp, 2025-09-09:
OS / Env
- Ubuntu 24.04.2 LTS (Docker, lsb_release -a)
- gz sim --version → 10.0.0~pre1
- Renderer: ogre2 (GZ_RENDER_ENGINE=ogre2), Qt: QT_QPA_PLATFORM=xcb

World
- video_record_dbl_pendulum.sdf from tutorial

Steps to reproduce
1. Reset GUI
```bash
rm -f ~/.gz/sim/10/gui.config
```
2. Start world
```bash
gz sim ~/worlds/video_record_dbl_pendulum.sdf
```
3. Open Video recorder on bottom right, choose any option
4. Click stop after recoding for any amount of time

Expected:
- After pressing stop, a dialogue for saving the recording should pop up

Actual
- No dialogue appears
- nothing was saved automatically
```bash
find / -type f \( -name "*.mp4" -o -name "*.ogv" \) -mtime -1 2>/dev/null | head
```
did not yield any results

Notes:
- terminal log only shows the start of the plugin, no debug messages such as "Video Recorder plugin is recording camera...", "saved to ..." etc.
- both formats fail to save any file
- tested starting from both ~ and / (if CWD is relevant)
- $HOME is writable
- GUI reset does not change behavior

[image]

[status: failed]