gazebosim/gazebo_test_cases #2253: gz-sim: visualize_lidar.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/visualize_lidar.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

akky20, 2025-08-27:
working on this

akky20, 2025-08-27:
Working fine ! No issues..

[image]

[image]

My system:
[image]

closed 2025-08-28 by ahcorde