gazebosim/gazebo_test_cases #2254: gz-sim: wide_angle_camera.sdf

azeey-test, 2025-08-27:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim10/examples/worlds/wide_angle_camera.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

pratik-adhikari, 2025-08-27:
[image]

[image]

**[status: passed]** - Wide angle camera distortion (180° FOV, tangent mapping) verified working in Jetty on Ubuntu 24.04 Jetty container - basic regression test confirms functionality remains intact

ahcorde, 2025-08-28:
Thank you for testing!

closed 2025-08-28 by ahcorde