gazebosim/gazebo_test_cases #2265: gz-transport: Installation

azeey-test, 2025-08-27:
## Configuration

* os: Ubuntu Noble
* transport: zenoh

## Links

* [Live](https://gazebosim.org/api/transport/15/installation.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport15/tutorials/02_installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
If the tutorial is an .sdf file, please open the file and read the comments at the top of the file.
        

    

    
    
            
### Step 2 - Images (if there are any) match the result
            
        

    
If the tutorial is an .sdf file, post a screenshot of the simulation run.
        

    

    
    
            
### Step 3 - Comment status
            
        

    
Include a special comment at the end of your post to indicate pass/fail. Use either `[status: passed]` or `[status: failed]`.

nikodemj9, 2025-08-27:
Binary seems to be already installed while installing gazebo.

Tried cloning and building from the repository, prerequisites mention installing python, but actually had to install python-is-python3 and python3, as python is no longer available in ubuntu repositories.

[image]

[status: failed]

caguero, 2025-08-27:
Thanks @nikodemj9 . I just created a patch to update the instructions. I tried it on Docker but could you give it a try?

https://github.com/gazebosim/gz-transport/pull/700